rcore-os/rCore-Tutorial-v3 #1: [Question] 执行 sbi shutdown 以后，后面的语句还能被执行吗？

mingyang91, 2021-01-13:
代码出自 https://github.com/rcore-os/rCore-Tutorial-v3/blob/ch1/os/src/sbi.rs#L37
```rust
pub fn shutdown() -> ! {
    sbi_call(SBI_SHUTDOWN, 0, 0, 0);
    panic!("It should shutdown!"); // <- 这里还能被执行到吗？
}
```

wyfcyx, 2021-01-13:
这里调用了 shutdown 的 sbi call，因此它的行为取决于底层 SBI 的实现。目前 RustSBI 的实现应该是，在 Qemu 平台上会通过某种方式通知 Qemu 自己直接退出；在 K210 平台上则会在 RustSBI 里面死循环（？）。所以一般情况下这里的 panic 是执行不到的。反过来说如果执行了这个 panic，说明一定有哪里出了问题（sbi 实现有误）。因此这个 panic 的作用是检查错误，它并不期待被执行到。

mingyang91, 2021-01-13:
明白了，原来是这个作用
谢谢

closed 2021-01-13 by wyfcyx

wangrunji0408, 2021-01-13:
所以这里用 `unreachable!()` 代替 `panic!()` 会更清楚些

wyfcyx, 2021-01-13:
> 所以这里用 `unreachable!()` 代替 `panic!()` 会更清楚些

之前是这样做的，但是 Rust 编译器似乎只会对应插入一条 unimp 指令，结果就是只能触发非法指令异常，不容易定位到底是哪里的 unreachable，所以就改成这样的实现了。

wangrunji0408, 2021-01-13:
啊这，我还以为这两个是一样的