rcore-os/rCore-Tutorial-v3 #4: error[E0557]: feature has been removed

Michael-Unix, 2021-02-02:
平台：WSL2
操作系统： Ubuntu18.04/Ubuntu20.04 
Rust版本：rustc 1.49.0

在每次rCore-Tutorial-v3 os中进行make run时，都会提示：
error[E0557]: feature has been removed
 --> src/main.rs:6:12
  |
6 | #![feature(const_in_array_repeat_expressions)]
  |            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ feature has been removed
  |
  = note: removed due to causing promotable bugs

error: aborting due to previous error

For more information about this error, try `rustc --explain E0557`.
error: could not compile `os`

wyfcyx, 2021-02-02:
在 os 目录下新建一个文件 rust-toolchain ，里面写上 ``nightly`` ，然后 ``rustup update`` 。我这里都更新到最新的 rustc 1.51 并没出现这个问题。另外这是哪个分支？

Michael-Unix, 2021-02-04:
* main

Zp121382, 2021-05-30:
> 在 os 目录下新建一个文件 rust-toolchain ，里面写上 `nightly` ，然后 `rustup update` 。我这里都更新到最新的 rustc 1.51 并没出现这个问题。另外这是哪个分支？

你好，我的环境是Ubuntu20.04，在os中加rust-toolchain这个文件了，但是make run BOARD=k210之后还是会报这样的错误

Warmchay, 2021-08-08:
- 操作系统： Ubuntu18.04/Ubuntu20.04
- rustc --version : `rustc 1.56.0-nightly (574d37568 2021-08-07)`
同样也在 os 目录下创建了 `rust-toolchain` 这个目录，没有理解写入 `nightly` 是什么意思，创建了一个 `nightly` 的文件
make run 后出现的问题是：
```
error[E0557]: feature has been removed
 --> src/main.rs:6:12
  |
6 | #![feature(const_in_array_repeat_expressions)]
  |            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ feature has been removed
  |
  = note: removed due to causing promotable bugs
```
是需要降低 rust 版本吗 ？

wangshangming, 2021-08-15:
可以把 `rCore-Tutorial-v3/rust-toolchain` 的内容改成 `nightly-2021-02-01-x86_64-unknown-linux-gnu`，我试了下可以

LimingFang, 2021-08-15:
我在mac上也遇到一模一样的问题。
rustc是rustc 1.56.0-nightly (8007b506a 2021-08-14)

> os git:(main) ✗ rustup show
Default host: x86_64-apple-darwin
rustup home:  /Users/limingfang/.rustup

installed toolchains
--------------------

stable-x86_64-apple-darwin
nightly-x86_64-apple-darwin (default)

installed targets for active toolchain
--------------------------------------

riscv64gc-unknown-none-elf
x86_64-apple-darwin

active toolchain
----------------

nightly-x86_64-apple-darwin (overridden by '/Users/limingfang/Documents/GitHub/cloned/rCore-Tutorial-v3/rust-toolchain')
rustc 1.56.0-nightly (8007b506a 2021-08-14)

LimingFang, 2021-08-15:
> 平台：WSL2
> 操作系统： Ubuntu18.04/Ubuntu20.04
> Rust版本：rustc 1.49.0
>
> 在每次rCore-Tutorial-v3 os中进行make run时，都会提示：
> error[E0557]: feature has been removed
> --> src/main.rs:6:12
> |
> 6 | [...]

我是Mac。

直接去`os/main.rs`下第六行把`#![feature(const_in_array_repeat_expressions)]`注释掉就work了，但是不知道这个是否可取。

Warmchay, 2021-08-20:
> * 操作系统： Ubuntu18.04/Ubuntu20.04
> * rustc --version : `rustc 1.56.0-nightly (574d37568 2021-08-07)`
>   同样也在 os 目录下创建了 `rust-toolchain` 这个目录，没有理解写入 `nightly` 是什么意思，创建了一个 `nightly` 的文件
>   make run [...]

在 `src/main.rs` 这一行已经用 `#` 注释了，我将这句删去后编译成功了 : )

rain-sumire, 2021-09-09:
遇到相同的问题了

> 在 `src/main.rs` 这一行已经用 `#` 注释了，我将这句删去后编译成功了 : )

pluveto, 2021-09-16:
同样的问题。版本：

```
rustc --version
rustc 1.57.0-nightly (2c7bc5e33 2021-09-15)
```

操作系统为 Elementary OS

 
@wangshangming 的方法无效
@LimingFang 的方法有效

BackMountainDevil, 2021-10-03:
# 问题描述

使用最新的 rustc 1.57.0-nightly (f03eb6bef 2021-10-02) 编译最新的仓库代码出现许多错误（大概是该参数已被废弃），随后看 issue 使用项目中的指定版本 nightly-2021-01-30 进行编译，仍然存在 3 个错误，信息如下

```bash
[kearney@xx rCore-Tutorial-v3]$ cd os/
[kearney@xx os]$ cargo build
   Compiling os v0.1.0 (/home/kearney/Documents/code/rust/rCore-Tutorial-v3/os)
error[E0432]: unresolved import `crate::config::CLOCK_FREQ`
 --> src/timer.rs:3:5
  |
3 | use crate::config::CLOCK_FREQ;
  |     ^^^^^^^^^^^^^^^^^^^^^^^^^ no `CLOCK_FREQ` in `config`

error[E0432]: unresolved import `crate::config::MMIO`
  --> src/mm/memory_set.rs:17:5
   |
17 |     MMIO,
   |     ^^^^ no `MMIO` in `config`

error[E0433]: failed to resolve: use of undeclared type `BlockDeviceImpl`
  --> src/drivers/block/mod.rs:15:66
   |
15 |     pub static ref BLOCK_DEVICE: Arc<dyn BlockDevice> = Arc::new(BlockDeviceImpl::new());
   |                                                                  ^^^^^^^^^^^^^^^ use of undeclared type `BlockDeviceImpl`

error: aborting due to 3 previous errors

Some errors have detailed explanations: E0432, E0433.
For more information about an error, try `rustc --explain E0432`.
error: could not compile `os`

To learn more, run the command again with --verbose.
```

# 版本信息

```bash
$ rustup show
Default host: x86_64-unknown-linux-gnu
rustup home:  /home/kearney/.rustup

installed toolchains
--------------------

nightly-2021-01-30-x86_64-unknown-linux-gnu (default)
nightly-x86_64-unknown-linux-gnu

installed targets for active toolchain
--------------------------------------

riscv64gc-unknown-none-elf
x86_64-unknown-linux-gnu

active toolchain
----------------

nightly-2021-01-30-x86_64-unknown-linux-gnu (overridden by '/home/kearney/Documents/code/rust/rCore-Tutorial-v3/rust-toolchain')
rustc 1.51.0-nightly (b12290861 2021-01-29)
```

wyfcyx, 2021-10-03:
@BackMountainDevil 请在`os/`下使用`make build`或`make run`而非直接`cargo build/run`。

BackMountainDevil, 2021-10-03:
很感谢在国庆假期收到回复，跑完回来立马开机试了一下

```bash
[kearney@xx rCore-Tutorial-v3]$ cd os/
[kearney@xx os]$ make build
(rustup target list | grep "riscv64gc-unknown-none-elf (installed)") || rustup target add riscv64gc-unknown-none-elf
riscv64gc-unknown-none-elf (installed)
cargo install cargo-binutils --vers ~0.2
    Updating `git://mirrors.ustc.edu.cn/crates.io-index` index
     Ignored package `cargo-binutils v0.2.0` is already installed, use --force to override
rustup component add rust-src
info: component 'rust-src' is up to date
rustup component add llvm-tools-preview
info: component 'llvm-tools-preview' for target 'x86_64-unknown-linux-gnu' is up to date
Platform: qemu
cp: 无法创建普通文件'src/linker.ld': 权限不够
make: *** [Makefile:59：kernel] 错误 1
```

权限不足一般加个 sudo,可是根据猜测这里不需要对系统目录做修改的 root 权限吧，我试一下加个 sudo

BackMountainDevil, 2021-10-03:
加上了 sudo 出现了预想中的问题，东西都装到 root 的用户目录下了

```bash
[kearney@xx os]$ sudo make build
(rustup target list | grep "riscv64gc-unknown-none-elf (installed)") || rustup target add riscv64gc-unknown-none-elf
riscv64gc-unknown-none-elf (installed)
cargo install cargo-binutils --vers ~0.2
    Updating `git://mirrors.ustc.edu.cn/crates.io-index` index
     Ignored package `cargo-binutils v0.2.0` is already installed, use --force to override
rustup component add rust-src
info: component 'rust-src' is up to date
rustup component add llvm-tools-preview
info: component 'llvm-tools-preview' for target 'x86_64-unknown-linux-gnu' is up to date
Platform: qemu
    Finished release [optimized] target(s) in 0.01s
make: rust-objcopy: 没有那个文件或目录
make: *** [Makefile:49：target/riscv64gc-unknown-none-elf/release/os.bin] 错误 127
```

wyfcyx, 2021-10-03:
@BackMountainDevil 应该全程不需要root权限的。最好使用一个非root用户，将`rCore-Tutorial-v3`目录放到`~`下面的子目录就好了。

BackMountainDevil, 2021-10-04:
@wyfcyx 确实如此，根据上面的错误信息我再用普通用户去编译就会出现权限问题，把target全删了之后用普通用户在 os 下编译还是出现了上面的错误

```
make: rust-objcopy: 没有那个文件或目录
make: *** [Makefile:49：target/riscv64gc-unknown-none-elf/release/os.bin] 错误 127
```

我以为是没有生成 os.bin,确实是没有生成，但我在其它工程下（cargo build --release）生成的二进制文件就是叫做 os,然后修改了makefile 的 `KERNEL_BIN := $(KERNEL_ELF).bin`（将后缀去掉），错误类型和上面一样，最后我发现问题的原因在于我这次没有把 `export PATH="$HOME/.cargo/bin:$PATH"` 写入`~/.bashrc`。 
我的问题解决了，十分感谢 @wyfcyx  的回复和交流。祝君假期愉快。

closed 2022-03-24 by wyfcyx