rcore-os/rCore-Tutorial-v3 #7: Get `Could not allocate dynamic translator buffer` on Ubuntu 20.04 with `make docker`

qiukun, 2021-02-28:
Any ideas?

Ubuntu 20.04 was installed by `multipass` as a vm.
Logs:
```
root@eternal-buck:/home/ubuntu/rCore-Tutorial-v3# cat /etc/issue
Ubuntu 20.04.2 LTS \n \l

root@eternal-buck:/home/ubuntu/rCore-Tutorial-v3# make docker
docker run --rm -it --mount type=bind,source=/home/ubuntu/rCore-Tutorial-v3,destination=/mnt dinghao188/rcore-tutorial
root@9f61a9655d71:/# cd mnt
root@9f61a9655d71:/mnt# cd os

root@9f61a9655d71:/mnt/os# make run
info: syncing channel updates for 'nightly-2021-01-30-x86_64-unknown-linux-gnu'
info: latest update on 2021-01-30, rust version 1.51.0-nightly (b12290861 2021-01-29)
.....

Could not allocate dynamic translator buffer
Makefile:67: recipe for target 'run-inner' failed
make: *** [run-inner] Error 1

root@e4fb5cbf909a:/mnt/os# qemu-system-riscv64 --version
QEMU emulator version 5.0.0
Copyright (c) 2003-2020 Fabrice Bellard and the QEMU Project developers
```

wyfcyx, 2021-02-28:
If your platform is Apple M1, you can ``make docker`` on macOS directly and then ``cd /mnt/os && make run``, docker itself is a kind of VM.

The error ``Could not allocate dynamic translator buffer`` comes from the imcompability of Apple M1 and qemu, see the discussions [here](https://github.com/rcore-os/rCore-Tutorial-Book-v3/issues/11).

qiukun, 2021-02-28:
@wyfcyx In fact, it's in a VM on an Intel Hackintosh. Could the incompatibility you mentioned affect a VM?

qiukun, 2021-02-28:
Tried to use Ubuntu 18.04 and get the same result. Comparing to the book, I've only noticed the difference in the version of rust. It's possible to specify the rust version? I'd like to try. If this doesn't work, I may switch the backend from `Hypervisor.framework` to VirtualBox.

ZhangHanDong, 2021-02-28:
@qiukun 

You can try to build a mirror locally on m1.

> make build_docker 

Then you can run the code directly on m1. Please see Dockerfile and Makefile for details.

In addition, if you have used multipass, why not use qemu 5.0 directly on ubuntu?

qiukun, 2021-02-28:
Tks. I’ve switch to the docker on Mac approach now. The bug is still strange, though. Maybe I could try native Ubuntu later.

Spxg, 2021-03-08:
Maybe there's not enough memory. This has happened to me in WSL because the memory is being used more and more by a VSCode plugin.