rcore-os/rCore-Tutorial-v3 #8: 使用docker环境时,make run报错,找不到glibc2.3

jimersylee, 2021-03-04:
已经按教程配置了docker环境,进入docker容器中,make run 报错,找不到glibc2.3

wyfcyx, 2021-03-04:
更详细的平台信息和错误截图？

jimersylee, 2021-03-05:
info: component 'llvm-tools-preview' for target 'x86_64-unknown-linux-gnu' is up to date
Platform: qemu
    Updating git repository `https://github.com/rcore-os/virtio-drivers`
  Downloaded bare-metal v0.2.5 (registry `git://mirrors.ustc.edu.cn/crates.io-index`)
  Downloaded buddy_system_allocator v0.6.0 (registry `git://mirrors.ustc.edu.cn/crates.io-index`)
  Downloaded embedded-hal v1.0.0-alpha.4 (registry `git://mirrors.ustc.edu.cn/crates.io-index`)
  Downloaded lazy_static v1.4.0 (registry `git://mirrors.ustc.edu.cn/crates.io-index`)
  Downloaded log v0.4.14 (registry `git://mirrors.ustc.edu.cn/crates.io-index`)
  Downloaded nb v0.1.3 (registry `git://mirrors.ustc.edu.cn/crates.io-index`)
  Downloaded nb v1.0.0 (registry `git://mirrors.ustc.edu.cn/crates.io-index`)
  Downloaded riscv v0.6.0 (registry `git://mirrors.ustc.edu.cn/crates.io-index`)
  Downloaded riscv-target v0.1.2 (registry `git://mirrors.ustc.edu.cn/crates.io-index`)
  Downloaded spin v0.5.2 (registry `git://mirrors.ustc.edu.cn/crates.io-index`)
  Downloaded spin v0.7.1 (registry `git://mirrors.ustc.edu.cn/crates.io-index`)
  Downloaded vcell v0.1.3 (registry `git://mirrors.ustc.edu.cn/crates.io-index`)
  Downloaded volatile v0.2.7 (registry `git://mirrors.ustc.edu.cn/crates.io-index`)
  Downloaded xmas-elf v0.7.0 (registry `git://mirrors.ustc.edu.cn/crates.io-index`)
  Downloaded zero v0.1.2 (registry `git://mirrors.ustc.edu.cn/crates.io-index`)
  Downloaded bit_field v0.10.1 (registry `git://mirrors.ustc.edu.cn/crates.io-index`)
  Downloaded 16 crates (253.1 KB) in 5.05s
   Compiling k210-pac v0.2.0 (https://github.com/wyfcyx/k210-pac#8f99c1ca)
   Compiling os v0.1.0 (/mnt/os)
   Compiling bitflags v1.2.1
   Compiling aho-corasick v0.7.15
   Compiling log v0.4.14
   Compiling bare-metal v0.2.5
error: failed to run custom build command for `os v0.1.0 (/mnt/os)`

Caused by:
  process didn't exit successfully: `/mnt/os/target/release/build/os-9c2bb3ea6a386156/build-script-build` (exit code: 1)
  --- stderr
  /mnt/os/target/release/build/os-9c2bb3ea6a386156/build-script-build: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /mnt/os/target/release/build/os-9c2bb3ea6a386156/build-script-build)
  /mnt/os/target/release/build/os-9c2bb3ea6a386156/build-script-build: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /mnt/os/target/release/build/os-9c2bb3ea6a386156/build-script-build)
warning: build failed, waiting for other jobs to finish...
error: build failed
Makefile:58: recipe for target 'kernel' failed
make: *** [kernel] Error 101

wyfcyx, 2021-03-05:
暂时不太清错误原因，可以试试手动配置环境。

closed 2021-10-07 by wyfcyx