rcore-os/rCore-Tutorial-v3 #11: 第五章的内核代码好像有点问题

ZimingYuan, 2021-03-31:
测试程序：
hello_world0.rs：
```rust
#![no_std]
#![no_main]

#[macro_use]
extern crate user_lib;

use user_lib::{fork, exec};

#[no_mangle]
pub fn main() -> i32 {
    let p = fork();
    for i in 0..10 {
        if p == 0 {
            println!("Program[0] print{}", i);
        } else {
            println!("Program[1] print{}", i);
        }
    }
    if p == 0 {
        exec("hello_world2");
    } else {
        exec("hello_world1");
    }
    0
}
```
hello_world1.rs：
```rust
#![no_std]
#![no_main]

#[macro_use]
extern crate user_lib;

#[no_mangle]
pub fn main() -> i32 {
    for i in 0..10 {
        println!("Program[1] print{}", i);
    }
    0
}
```
hello_world2.rs：
```rust
#![no_std]
#![no_main]

#[macro_use]
extern crate user_lib;

#[no_mangle]
pub fn main() -> i32 {
    for i in 0..10 {
        println!("Program[2] print{}", i);
    }
    0
}
```
第五章分支下的rCore运行hello_world0只输出了调用exec前应该输出的内容，没有输出exec后hello_world1和hello_world2应该输出的内容，不清楚是什么原因。

wyfcyx, 2021-03-31:
@ZimingYuan 内核没有问题，实际上是 ``hello_world0`` 里面的两个 ``exec`` 执行均因为找不到应用而失败，返回值为``-1``并继续执行原应用。在应用里面手动 ``exec`` 的时候，应用名的最后要有一个 `\0` ，如 ``exec("hello_world1\0")``，即可得到期望输出。

ZimingYuan, 2021-03-31:
感谢！C语言写惯了没有注意到Rust和C常量字符串的区别。

closed 2021-03-31 by ZimingYuan