rcore-os/rCore-Tutorial-v3 #13: build fail

oska874, 2021-04-03:
```
   Compiling riscv v0.6.0
   Compiling riscv v0.6.0 (https://github.com/rcore-os/riscv#0074cbc9)
   Compiling riscv v0.6.0 (https://github.com.cnpmjs.org/rcore-os/riscv#0074cbc9)
   Compiling k210-hal v0.2.0 (https://github.com/wyfcyx/k210-hal#ea121c91)
   Compiling k210-hal v0.2.0 (https://github.com.cnpmjs.org/wyfcyx/k210-hal#ea121c91)
   Compiling k210-soc v0.1.0 (https://github.com.cnpmjs.org/wyfcyx/k210-soc#09a0742c)
error[E0599]: the method `configure` exists for struct `k210_pac::UARTHS`, but its trait bounds were not satisfied
    --> src/drivers/block/sdcard.rs:725:24
     |
725  |     peripherals.UARTHS.configure(115_200.bps(), &clocks);
     |                        ^^^^^^^^^ method cannot be called on `k210_pac::UARTHS` due to unsatisfied trait bounds
     | 
    ::: /home/zhang/.cargo/git/checkouts/k210-pac-34eac973c22ba7ac/8f99c1c/src/lib.rs:1028:1
     |
1028 | pub struct UARTHS {
     | -----------------
     | |
     | doesn't satisfy `k210_pac::UARTHS: SerialExt`
     | doesn't satisfy `k210_pac::UARTHS: serial::closed_trait::UartX`
...
1050 |     pub struct RegisterBlock {
     |     ------------------------
     |     |
     |     doesn't satisfy `_: serial::closed_trait::UartX`
     |     doesn't satisfy `k210_pac::uarths::RegisterBlock: SerialExt`
     |
     = note: the following trait bounds were not satisfied:
             `k210_pac::UARTHS: serial::closed_trait::UartX`
             which is required by `k210_pac::UARTHS: SerialExt`
             `&k210_pac::UARTHS: serial::closed_trait::UartX`
             which is required by `&k210_pac::UARTHS: SerialExt`
             `&mut k210_pac::UARTHS: serial::closed_trait::UartX`
             which is required by `&mut k210_pac::UARTHS: SerialExt`
             `k210_pac::uarths::RegisterBlock: serial::closed_trait::UartX`
             which is required by `k210_pac::uarths::RegisterBlock: SerialExt`
             `&k210_pac::uarths::RegisterBlock: serial::closed_trait::UartX`
             which is required by `&k210_pac::uarths::RegisterBlock: SerialExt`
             `&mut k210_pac::uarths::RegisterBlock: serial::closed_trait::UartX`
             which is required by `&mut k210_pac::uarths::RegisterBlock: SerialExt`

error[E0599]: the method `constrain` exists for struct `k210_pac::SPI0`, but its trait bounds were not satisfied
     --> src/drivers/block/sdcard.rs:728:32
      |
728   |     let spi = peripherals.SPI0.constrain();
      |                                ^^^^^^^^^ method cannot be called on `k210_pac::SPI0` due to unsatisfied trait bounds
      | 
     ::: /home/zhang/.cargo/git/checkouts/k210-pac-34eac973c22ba7ac/8f99c1c/src/lib.rs:19849:1
      |
19849 | pub struct SPI0 {
      | ---------------
      | |
      | doesn't satisfy `k210_pac::SPI0: SPI01`
      | doesn't satisfy `k210_pac::SPI0: SPIExt`
...
19871 |     pub struct RegisterBlock {
      |     ------------------------
      |     |
      |     doesn't satisfy `k210_pac::spi0::RegisterBlock: SPI01`
      |     doesn't satisfy `k210_pac::spi0::RegisterBlock: SPIExt`
      |
      = note: the following trait bounds were not satisfied:
              `k210_pac::SPI0: SPI01`
              which is required by `k210_pac::SPI0: SPIExt`
              `&k210_pac::SPI0: SPI01`
              which is required by `&k210_pac::SPI0: SPIExt`
              `&mut k210_pac::SPI0: SPI01`
              which is required by `&mut k210_pac::SPI0: SPIExt`
              `k210_pac::spi0::RegisterBlock: SPI01`
              which is required by `k210_pac::spi0::RegisterBlock: SPIExt`
              `&k210_pac::spi0::RegisterBlock: SPI01`
              which is required by `&k210_pac::spi0::RegisterBlock: SPIExt`
              `&mut k210_pac::spi0::RegisterBlock: SPI01`
              which is required by `&mut k210_pac::spi0::RegisterBlock: SPIExt`

error[E0277]: the trait bound `k210_pac::SPI0: SPI01` is not satisfied
   --> src/drivers/block/sdcard.rs:730:19
    |
730 |     let info = sd.init().unwrap();
    |                   ^^^^ the trait `SPI01` is not implemented for `k210_pac::SPI0`
    |
    = note: required because of the requirements on the impl of `SPI` for `SPIImpl<k210_pac::SPI0>`

error[E0277]: the trait bound `k210_pac::SPI0: SPI01` is not satisfied
   --> src/drivers/block/sdcard.rs:729:14
    |
163 |     pub fn new(spi: X, spi_cs: u32, cs_gpionum: u8/*, dmac: &'a DMAC, channel: dma_channel*/) -> Self {
    |     ------------------------------------------------------------------------------------------------- required by `SDCard::<X>::new`
...
729 |     let sd = SDCard::new(spi, SD_CS, SD_CS_GPIONUM);
    |              ^^^^^^^^^^^ the trait `SPI01` is not implemented for `k210_pac::SPI0`
    |
    = note: required because of the requirements on the impl of `SPI` for `SPIImpl<k210_pac::SPI0>`

error[E0599]: the method `read_sector` exists for struct `MutexGuard<'_, SDCard<SPIImpl<k210_pac::SPI0>>>`, but its trait bounds were not satisfied
   --> src/drivers/block/sdcard.rs:748:23
    |
748 |         self.0.lock().read_sector(buf,block_id as u32).unwrap();
    |                       ^^^^^^^^^^^ method cannot be called on `MutexGuard<'_, SDCard<SPIImpl<k210_pac::SPI0>>>` due to unsatisfied trait bounds
    | 
   ::: /home/zhang/.cargo/git/checkouts/k210-soc-2d953ceb9d03629e/09a0742/src/spi.rs:52:1
    |
52  | pub struct SPIImpl<IF> {
    | ---------------------- doesn't satisfy `SPIImpl<k210_pac::SPI0>: SPI`
    |
    = note: the following trait bounds were not satisfied:
            `SPIImpl<k210_pac::SPI0>: SPI`

error[E0599]: the method `write_sector` exists for struct `MutexGuard<'_, SDCard<SPIImpl<k210_pac::SPI0>>>`, but its trait bounds were not satisfied
   --> src/drivers/block/sdcard.rs:751:23
    |
751 |         self.0.lock().write_sector(buf,block_id as u32).unwrap();
    |                       ^^^^^^^^^^^^ method cannot be called on `MutexGuard<'_, SDCard<SPIImpl<k210_pac::SPI0>>>` due to unsatisfied trait bounds
    | 
   ::: /home/zhang/.cargo/git/checkouts/k210-soc-2d953ceb9d03629e/09a0742/src/spi.rs:52:1
    |
52  | pub struct SPIImpl<IF> {
    | ---------------------- doesn't satisfy `SPIImpl<k210_pac::SPI0>: SPI`
    |
    = note: the following trait bounds were not satisfied:
            `SPIImpl<k210_pac::SPI0>: SPI`

error: aborting due to 6 previous errors

Some errors have detailed explanations: E0277, E0599.
For more information about an error, try `rustc --explain E0277`.
error: could not compile `os`

To learn more, run the command again with --verbose.
make: *** [Makefile:60: kernel] Error 101
```
rCore-Tutorial-v3 version is 60477da9be6879cdfa88e24c8c5af2f353117ac8.

wyfcyx, 2021-04-03:
I have tried in my environment but it worked as usual. Could you please give us some infomation about your environment? By the way, why did you fetch k210 crates from `github.com.cnpmjs.org`? It sounds weird.

wyfcyx, 2021-04-23:
Switch rustc to nightly, see #17 .

closed 2021-04-23 by wyfcyx