rcore-os/rCore-Tutorial-v3 #17: the method `configure` exists for struct `k210_pac::UARTHS`, but its trait bounds were not satisfied

wangzhankun, 2021-04-23:
当我参考 https://rcore-os.github.io/rCore-Tutorial-Book-v3/chapter0/5setup-devel-env.html 配置环境完成之后进行编译，报错如下：
```bash
error[E0599]: the method `configure` exists for struct `k210_pac::UARTHS`, but its trait bounds were not satisfied
    --> src/drivers/block/sdcard.rs:725:24
     |
725  |     peripherals.UARTHS.configure(115_200.bps(), &clocks);
     |                        ^^^^^^^^^ method cannot be called on `k210_pac::UARTHS` due to unsatisfied trait bounds
     |
    ::: /home/wang/.cargo/git/checkouts/k210-pac-19d36f600eb54ac3/8f99c1c/src/lib.rs:1028:1
     |
1028 | pub struct UARTHS {
     | -----------------
     | |
     | doesn't satisfy `k210_pac::UARTHS: SerialExt`
     | doesn't satisfy `k210_pac::UARTHS: serial::closed_trait::UartX`
...
1050 |     pub struct RegisterBlock {
     |     ------------------------
     |     |
     |     doesn't satisfy `_: serial::closed_trait::UartX`
     |     doesn't satisfy `k210_pac::uarths::RegisterBlock: SerialExt`
     |
     = note: the following trait bounds were not satisfied:
             `k210_pac::UARTHS: serial::closed_trait::UartX`
             which is required by `k210_pac::UARTHS: SerialExt`
             `&k210_pac::UARTHS: serial::closed_trait::UartX`
             which is required by `&k210_pac::UARTHS: SerialExt`
             `&mut k210_pac::UARTHS: serial::closed_trait::UartX`
             which is required by `&mut k210_pac::UARTHS: SerialExt`
             `k210_pac::uarths::RegisterBlock: serial::closed_trait::UartX`
             which is required by `k210_pac::uarths::RegisterBlock: SerialExt`
             `&k210_pac::uarths::RegisterBlock: serial::closed_trait::UartX`
             which is required by `&k210_pac::uarths::RegisterBlock: SerialExt`
             `&mut k210_pac::uarths::RegisterBlock: serial::closed_trait::UartX`
             which is required by `&mut k210_pac::uarths::RegisterBlock: SerialExt`

error[E0599]: the method `constrain` exists for struct `k210_pac::SPI0`, but its trait bounds were not satisfied
     --> src/drivers/block/sdcard.rs:728:32
      |
728   |     let spi = peripherals.SPI0.constrain();
      |                                ^^^^^^^^^ method cannot be called on `k210_pac::SPI0` due to unsatisfied trait bounds
      |
     ::: /home/wang/.cargo/git/checkouts/k210-pac-19d36f600eb54ac3/8f99c1c/src/lib.rs:19849:1
      |
19849 | pub struct SPI0 {
      | ---------------
      | |
      | doesn't satisfy `k210_pac::SPI0: SPI01`
      | doesn't satisfy `k210_pac::SPI0: SPIExt`
...
19871 |     pub struct RegisterBlock {
      |     ------------------------
      |     |
      |     doesn't satisfy `k210_pac::spi0::RegisterBlock: SPI01`
      |     doesn't satisfy `k210_pac::spi0::RegisterBlock: SPIExt`
      |
      = note: the following trait bounds were not satisfied:
              `k210_pac::SPI0: SPI01`
              which is required by `k210_pac::SPI0: SPIExt`
              `&k210_pac::SPI0: SPI01`
              which is required by `&k210_pac::SPI0: SPIExt`
              `&mut k210_pac::SPI0: SPI01`
              which is required by `&mut k210_pac::SPI0: SPIExt`
              `k210_pac::spi0::RegisterBlock: SPI01`
              which is required by `k210_pac::spi0::RegisterBlock: SPIExt`
              `&k210_pac::spi0::RegisterBlock: SPI01`
              which is required by `&k210_pac::spi0::RegisterBlock: SPIExt`
              `&mut k210_pac::spi0::RegisterBlock: SPI01`
              which is required by `&mut k210_pac::spi0::RegisterBlock: SPIExt`

error[E0277]: the trait bound `k210_pac::SPI0: SPI01` is not satisfied
   --> src/drivers/block/sdcard.rs:730:19
    |
730 |     let info = sd.init().unwrap();
    |                   ^^^^ the trait `SPI01` is not implemented for `k210_pac::SPI0`
    |
    = note: required because of the requirements on the impl of `SPI` for `SPIImpl<k210_pac::SPI0>`

error[E0277]: the trait bound `k210_pac::SPI0: SPI01` is not satisfied
   --> src/drivers/block/sdcard.rs:729:14
    |
163 |     pub fn new(spi: X, spi_cs: u32, cs_gpionum: u8/*, dmac: &'a DMAC, channel: dma_channel*/) -> Self {
    |     ------------------------------------------------------------------------------------------------- required by `SDCard::<X>::new`
...
729 |     let sd = SDCard::new(spi, SD_CS, SD_CS_GPIONUM);
    |              ^^^^^^^^^^^ the trait `SPI01` is not implemented for `k210_pac::SPI0`
    |
    = note: required because of the requirements on the impl of `SPI` for `SPIImpl<k210_pac::SPI0>`

error[E0599]: the method `read_sector` exists for struct `MutexGuard<'_, SDCard<SPIImpl<k210_pac::SPI0>>>`, but its trait bounds were not satisfied
   --> src/drivers/block/sdcard.rs:748:23
    |
748 |         self.0.lock().read_sector(buf,block_id as u32).unwrap();
    |                       ^^^^^^^^^^^ method cannot be called on `MutexGuard<'_, SDCard<SPIImpl<k210_pac::SPI0>>>` due to unsatisfied trait bounds
    |
   ::: /home/wang/.cargo/git/checkouts/k210-soc-069e4fc1d9165bc2/09a0742/src/spi.rs:52:1
    |
52  | pub struct SPIImpl<IF> {
    | ---------------------- doesn't satisfy `SPIImpl<k210_pac::SPI0>: SPI`
    |
    = note: the following trait bounds were not satisfied:
            `SPIImpl<k210_pac::SPI0>: SPI`

error[E0599]: the method `write_sector` exists for struct `MutexGuard<'_, SDCard<SPIImpl<k210_pac::SPI0>>>`, but its trait bounds were not satisfied
   --> src/drivers/block/sdcard.rs:751:23
    |
751 |         self.0.lock().write_sector(buf,block_id as u32).unwrap();
    |                       ^^^^^^^^^^^^ method cannot be called on `MutexGuard<'_, SDCard<SPIImpl<k210_pac::SPI0>>>` due to unsatisfied trait bounds
    |
   ::: /home/wang/.cargo/git/checkouts/k210-soc-069e4fc1d9165bc2/09a0742/src/spi.rs:52:1
    |
52  | pub struct SPIImpl<IF> {
    | ---------------------- doesn't satisfy `SPIImpl<k210_pac::SPI0>: SPI`
    |
    = note: the following trait bounds were not satisfied:
            `SPIImpl<k210_pac::SPI0>: SPI`

error: aborting due to 6 previous errors

Some errors have detailed explanations: E0277, E0599.
For more information about an error, try `rustc --explain E0277`.
error: could not compile `os`

To learn more, run the command again with --verbose.
make: *** [Makefile:60: kernel] Error 101
```

wyfcyx, 2021-04-23:
请问你的环境是？我在Ubuntu虚拟机和虚拟机上的docker均能正常构建。

wangzhankun, 2021-04-23:
WSL Ubuntu 20.04

wyfcyx, 2021-04-23:
如果在k210上跑的话最好还是选择Ubuntu虚拟机，我们目前没有在WSL上做过测试:(

wangzhankun, 2021-04-23:
发现是rustc的版本问题，我更新到nightly之后就好了。谢谢。

closed 2021-04-23 by wangzhankun