rcore-os/rCore-Tutorial-v3 #18: [rustsbi-panic] system shutdown scheduled due to RustSBI panic

BackMountainDevil, 2021-05-28:
# 个人环境

```bash
$ uname -a
Linux arch 5.12.5-arch1-1 #1 SMP PREEMPT Wed, 19 May 2021 10:32:40 +0000 x86_64 GNU/Linux

$ rustc --version
rustc 1.54.0-nightly (1c6868aa2 2021-05-27)

$ qemu-riscv64 -version
qemu-riscv64 version 6.0.0

$ qemu-system-riscv64 -version
QEMU emulator version 6.0.0
```
配置过程参考了[文档 第零章：操作系统概述 » 实验环境配置](https://rcore-os.github.io/rCore-Tutorial-Book-v3/chapter0/5setup-devel-env.html#qemu)

rust 配置了nightly 版本，相应的工具已经安装（riscv64gc-unknown-none-elf, cargo-binutils --vers ~0.2, llvm-tools-preview, rust-src），有一个小小的问题是 那个 cargo-binutils 的版本一定要是 0.2 吗？虽然不知道为什么，但我还是按照文档的要求装了0.2 版的

qemu-riscv64 的版本比文档的高，文档为 5.0.0

# 错误描述
## 重现步骤

```bash
$ git clone git@github.com:rcore-os/rCore-Tutorial-v3.git
$ cd rCore-Tutorial-v3/os
$ make run
```

## 错误输出

```bash
known-linux-gnu (installed)
[kearney@arch os]$ make run
(rustup target list | grep "riscv64gc-unknown-none-elf (installed)") || rustup target add riscv64gc-unknown-none-elf
riscv64gc-unknown-none-elf (installed)
cargo install cargo-binutils --vers ~0.2
    Updating `https://mirrors.tuna.tsinghua.edu.cn/git/crates.io-index.git` index
     Ignored package `cargo-binutils v0.2.0` is already installed, use --force to override
rustup component add rust-src
info: component 'rust-src' is up to date
rustup component add llvm-tools-preview
info: component 'llvm-tools-preview' for target 'x86_64-unknown-linux-gnu' is up to date
Platform: qemu
    Finished release [optimized] target(s) in 0.01s
[rustsbi] RustSBI version 0.2.0-alpha.1
.______       __    __      _______.___________.  _______..______   __
|   _  \     |  |  |  |    /       |           | /       ||   _  \ |  |
|  |_)  |    |  |  |  |   |   (----`---|  |----`|   (----`|  |_)  ||  |
|      /     |  |  |  |    \   \       |  |      \   \    |   _  < |  |
|  |\  \----.|  `--'  |.----)   |      |  |  .----)   |   |  |_)  ||  |
| _| `._____| \______/ |_______/       |__|  |_______/    |______/ |__|

[rustsbi] Platform: QEMU (Version 0.2.0)
[rustsbi] misa: RV64ACDFIMSU
[rustsbi] mideleg: 0x222
[rustsbi] medeleg: 0xb1ab
[rustsbi-dtb] Hart count: cluster0 with 1 cores
[rustsbi] Kernel entry: 0x80200000
[rustsbi-panic] hart 0 panicked at 'invalid instruction, mepc: 000000008000004c, instruction: 0000000030200073', platform/qemu/src/main.rs:484:17
[rustsbi-panic] system shutdown scheduled due to RustSBI panic
```

# 正常的结果

在[文档里](https://rcore-os.github.io/rCore-Tutorial-Book-v3/chapter0/5setup-devel-env.html#rcore-tutorial-v3)的 GIF 图显示 make run 之后会加载内核，进入用户 shell

# 期望的结果

正常加载内核，进入 shell

BackMountainDevil, 2021-05-28:
使用 Docker version 20.10.6 下的 qemu 5.0.0 启动成功，怀疑是自己的 qemu 版本高了不适配 rust spi,降级试一试

Arstman, 2021-05-29:
这个确实是qemu版本过高的问题, 我manjaro系统, 出现一样的问题.

不过我的解决方案是根据教程上下载qemu 5.00 自己编译, 并放入path里面;  编译的时候会报错缺一些lib,  主要是缺riscv gun gcc的库, 和glusterfs, 安装好即可.

顺便说下, 要先把系统安装的qemu及其相关组件全部删掉才行.

BackMountainDevil, 2021-05-29:
# pacman 安装

对出现的 lib 问题都进行了动态链接，也补上了你说的 glusterfs,最后也没有成功。。请教下大佬说的 aur 里有的是指？

```bash
# 卸载当前的版本
$ sudo pacman -R qemu qemu-arch-extra

# 从本地安装包安装旧版的包
$ sudo pacman -U qemu-5.0.0-2-x86_64.pkg.tar.zst
$ sudo pacman -U qemu-arch-extra-5.0.0-2-x86_64.pkg.tar.zst

$ qemu-riscv64 -version
qemu-riscv64 version 5.0.0

$ qemu-system-riscv64 -version
qemu-system-riscv64: error while loading shared libraries: libbrlapi.so.0.7: cannot open shared object file: No such file or directory

# 创建软链接文件 libbrlapi.so.0.7
$ sudo ln -s libbrlapi.so.0.8.2 libbrlapi.so.0.7
# 对出现的同类 error 如法炮制
$ sudo ln -s liburing.so.2.0.0 liburing.so.1

$ sudo pacman -S glusterfs

$ qemu-system-riscv64 -version
qemu-system-riscv64: /usr/lib/liburing.so.1: version `LIBURING_0.2' not found (required by qemu-system-riscv64)
qemu-system-riscv64: /usr/lib/liburing.so.1: version `LIBURING_0.1' not found (required by qemu-system-riscv64)
```

# 编译安装

configure 没有 warning 或者  error 输出  
make 生成了不少 警告 和 附注，也没有 错误（ERROR）项，但是却中断了。。。  
log_make.txt 中 也没有  warning 或者  error ，全是 GEN, CC    
伤心太平洋

```bash
$ wget https://download.qemu.org/qemu-5.0.0.tar.xz
# 解压
$ tar xvJf qemu-5.0.0.tar.xz
$ cd qemu-5.0.0
# 配置 RISC-V 支持
$ ./configure --target-list=riscv64-softmmu,riscv64-linux-user
# 此处没有 warning 或者  error 输出
# 多线程编译
$ make -j$(nproc)>log_make.txt
ar: 正在创建 /home/kearney/Downloads/qemu-5.0.0/capstone/libcapstone.a
block/vpc.c: 在函数‘vpc_open’中:
block/vpc.c:360:51: 警告：array subscript ‘VHDDynDiskHeader {或称 struct vhd_dyndisk_header}[0]’ is partly outside array bounds of ‘uint8_t[512]’ {或称 ‘unsigned char[512]’} [-Warray-bounds]
  360 |         s->block_size = be32_to_cpu(dyndisk_header->block_size);
      |                                                   ^~
block/vpc.c:223:13: 附注：while referencing ‘buf’
  223 |     uint8_t buf[HEADER_SIZE];
      |             ^~~
block/vpc.c:368:58: 警告：array subscript ‘VHDDynDiskHeader {或称 struct vhd_dyndisk_header}[0]’ is partly outside array bounds of ‘uint8_t[512]’ {或称 ‘unsigned char[512]’} [-Warray-bounds]
  368 |         s->max_table_entries = be32_to_cpu(dyndisk_header->max_table_entries);
      |                                                          ^~
block/vpc.c:223:13: 附注：while referencing ‘buf’
  223 |     uint8_t buf[HEADER_SIZE];
      |             ^~~
block/vpc.c:400:51: 警告：array subscript ‘VHDDynDiskHeader {或称 struct vhd_dyndisk_header}[0]’ is partly outside array bounds of ‘uint8_t[512]’ {或称 ‘unsigned char[512]’} [-Warray-bounds]
  400 |         s->bat_offset = be64_to_cpu(dyndisk_header->table_offset);
      |                                                   ^~
block/vpc.c:223:13: 附注：while referencing ‘buf’
  223 |     uint8_t buf[HEADER_SIZE];
      |             ^~~
block/sheepdog.c: 在函数‘reload_inode’中:
block/sheepdog.c:1462:14: 警告：array subscript ‘SheepdogInode[0]’ is partly outside array bounds of ‘unsigned char[4664]’ [-Warray-bounds]
 1462 |     if (inode->vdi_id != s->inode.vdi_id) {
[... 17 lines cut ...]
block/sheepdog.c:2985:48: 警告：array subscript ‘SheepdogInode[0]’ is partly outside array bounds of ‘unsigned char[4664]’ [-Warray-bounds]
 2985 |             sn_tab[found].vm_clock_nsec = inode->vm_clock_nsec;
      |                                                ^~
block/sheepdog.c:2929:13: 附注：referencing an object of size 4664 allocated by ‘g_malloc’
 2929 |     inode = g_malloc(SD_INODE_HEADER_SIZE);
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
block/sheepdog.c:2988:39: 警告：array subscript ‘SheepdogInode[0]’ is partly outside array bounds of ‘unsigned char[4664]’ [-Warray-bounds]
 2988 |                      "%" PRIu32, inode->snap_id);
      |                                       ^~
block/sheepdog.c:2929:13: 附注：referencing an object of size 4664 allocated by ‘g_malloc’
 2929 |     inode = g_malloc(SD_INODE_HEADER_SIZE);
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
^Cmake: *** [/home/kearney/Downloads/qemu-5.0.0/rules.mak:69: blockdev.o] 中断
make: *** [/home/kearney/Downloads/qemu-5.0.0/rules.mak:69: fsdev/virtfs-proxy-helper.o] 中断
make: *** [/home/kearney/Downloads/qemu-5.0.0/rules.mak:69: contrib/vhost-user-gpu/vhost-user-gpu.o] 中断
make: *** [/home/kearney/Downloads/qemu-5.0.0/rules.mak:69: contrib/libvhost-user/libvhost-user.o] 中断
make: *** [/home/kearney/Downloads/qemu-5.0.0/rules.mak:69: tools/virtiofsd/buffer.o] 中断
make: *** [/home/kearney/Downloads/qemu-5.0.0/rules.mak:69: tools/virtiofsd/fuse_opt.o] 中断
make: *** [/home/kearney/Downloads/qemu-5.0.0/rules.mak:69: tools/virtiofsd/fuse_log.o] 中断
make: *** [/home/kearney/Downloads/qemu-5.0.0/rules.mak:69: tools/virtiofsd/fuse_lowlevel.o] 中断
make: *** [/home/kearney/Downloads/qemu-5.0.0/rules.mak:69: tools/virtiofsd/fuse_signals.o] 中断
make: *** [/home/kearney/Downloads/qemu-5.0.0/rules.mak:69: tools/virtiofsd/fuse_virtio.o] 中断
make: *** [/home/kearney/Downloads/qemu-5.0.0/rules.mak:69: tools/virtiofsd/helper.o] 中断
make: *** [/home/kearney/Downloads/qemu-5.0.0/rules.mak:69: tools/virtiofsd/passthrough_ll.o] 中断
```

Arstman, 2021-05-30:
为啥要对lib 问题都进行了动态链接?  这点我有点不理解. 我是没有这么操作的, 一般不建议这样把新版lib ln成旧版的.

aur  是Arch系的软件库啊,  相当于ubunut的软件源, 这可是是Linux世界里最大最全最新的软件库了, 你没有添加么?

BackMountainDevil, 2021-05-30:
emm有添加可以从 AUR 下载和降级的 yay ，downgrade, 但是用 downgrade 下 5.0.0 的qemu还是存在 lib 问题。缺失的 lib 除了补上动态链接之外，不行之后我就删掉了。 然后我尝试的是安装旧版的 lib对于的包（brltty、liburing），但是安装这些包就会要求这些包的依赖也同时要安装旧版的，就很头大。。

然后第三种办法，到网上找 lib文件

```bash
$ ldd /usr/bin/qemu-system-riscv64
libbrlapi.so.0.7 => not found
liburing.so.1 => not found
# libbrlapi.so 对于的包是 brltty
# liburing 对于的包是 liburing

# https://git.kernel.dk/cgit/liburing/ 下载0.1版的解压
$ cd liburing-0.1
./configure
make
sudo make install

# https://ubuntu.pkgs.org/20.04/ubuntu-main-amd64/libbrlapi0.7_6.0+dfsg-4ubuntu6_amd64.deb.html
# 解压deb,得到so文件。deb/data//lib/x86_64-linux-gnu
$ sudo cp libbrlapi.so.0.7 /usr/lib64/
$ sudo cp libbrlapi.so.0.7.0 /usr/lib64/
$ qemu-system-riscv64 -version
qemu-system-riscv64: /usr/lib/liburing.so.1: version `LIBURING_0.2` not found (required by qemu-system-riscv64)

# 系统中本来就是有liburing.so.2
$ sudo pacman -S liburing
$ qemu-system-riscv64 -version
qemu-system-riscv64: /usr/lib/liburing.so.1: version `LIBURING_0.2` not found (required by qemu-system-riscv64)
```

麻了麻了，清除继续寻找新方法，不知道大佬有何高见，望不吝赐教。

sakurafisch, 2021-06-28:
我今天也遇到了这个问题，着手解决了一下。
对软件降级可以使用 `downgrade` 这个工具，请查看这个 AUR 包哦 https://aur.archlinux.org/packages/downgrade/
```zsh
sudo pacman -S downgrade
sudo downgrade qemu
sudo downgrade qemu-arch-extra
```
这里需要手动选择一下要降到哪个版本。
其中 `qemu` 降到 `5.0.0` 版本，而 `qemu-arch-extra` 降到 `5.1.0` 版本。
这个方案在我的 Arch Linux 有效。

BackMountainDevil, 2021-06-29:
# qemu
这个办法我也尝试了。。。似乎在我这里有点问题，不知道你使用的arch 版本是最近几版吗？

```bash
sudo downgrade qemu-arch-extra
   24)  qemu-arch-extra    5.1.0  2  /var/cache/pacman/pkg
默认附带安装qemu6
sudo downgrade qemu
-  15)  qemu    5.0.0  2  /var/cache/pacman/pkg
qemu: 文件系统中存在 /usr/share/qemu/qemu-nsis.bmp （由 qemu-arch-extra 所有）
发生错误，没有软件包被更新。
```

反过来先装qemu在装risc

```bash
$ sudo downgrade qemu
-  15)  qemu    5.0.0  2  /var/cache/pacman/pkg
$ sudo downgrade qemu-arch-extra
-  24)  qemu-arch-extra    5.1.0  2  /var/cache/pacman/pkg
错误：无法提交处理 (有冲突的文件)
qemu-arch-extra: 文件系统中存在 /usr/share/qemu/qemu-nsis.bmp （由 qemu 所有）
发生错误，没有软件包被更新。
```

# os

Linux arch 5.12.10-arch1-1 #1 SMP PREEMPT Thu, 10 Jun 2021 16:34:50 +0000 x86_64 GNU/Linux

sakurafisch, 2021-06-29:
> # qemu
> 这个办法我也尝试了。。。似乎在我这里有点问题，不知道你使用的arch 版本是最近几版吗？
> 
> ```shell
> sudo downgrade qemu-arch-extra
>    24)  qemu-arch-extra    5.1.0  2  /var/cache/pacman/pkg
> 默认附带安装qemu6
> sudo downgrade qemu
> -  15)  qemu    5.0.0  2  /var/cache/pacman/pkg
> qemu: 文件系统中存在 /usr/share/qemu/qemu-nsis.bmp （由 qemu-arch-extra 所有）
> 发生错误，没有软件包被更新。
> ```
> 
> 反过来先装qemu在装risc
> 
> ```shell
> $ sudo downgrade qemu
> -  15)  qemu    5.0.0  2  /var/cache/pacman/pkg
> $ sudo downgrade qemu-arch-extra
> -  24)  qemu-arch-extra    5.1.0  2  /var/cache/pacman/pkg
> 错误：无法提交处理 (有冲突的文件)
> qemu-arch-extra: 文件系统中存在 /usr/share/qemu/qemu-nsis.bmp （由 qemu 所有）
> 发生错误，没有软件包被更新。
> ```
> 
> # os
> Linux arch 5.12.10-arch1-1 #1 SMP PREEMPT Thu, 10 Jun 2021 16:34:50 +0000 x86_64 GNU/Linux

更详细的信息
## OS
Linux archlinux 5.12.13-arch1-2 #1 SMP PREEMPT Fri, 25 Jun 2021 22:56:51 +0000 x86_64 GNU/Linux
## 降级的具体版本
```zsh
$ sudo downgrade qemu-arch-extra # 我选了 5.1.0 版本的最下方的一个
- 25) qemu-arch-extra    5.1.0  3  /var/cache/pacman/pkg
$ sudo downgrade qemu # 我选了 5.0.0 版本的最下方的一个
- 22) qemu    5.0.0  8  /var/cache/pacman/pkg
```

BackMountainDevil, 2021-06-29:
我os下午滚动更新了，没立即重启，重启完我两版本一致了。
按照你给的版本信息进行安装

- 25) qemu-arch-extra    5.1.0  3
- 21)  qemu    5.0.0  8  远端
```bash
软件包 (14) dtc-1.6.0-3  libcacard-2.7.0-2  libnfs-4.0.0-4  libslirp-4.6.1-1  liburing-2.0-1
            qemu-git-13:6.0.0.r1992.ge3955ae93f-1  seabios-1.14.0-1  spice-0.14.3-3  usbredir-0.9.0-1  vde2-2.3.2-16
            virglrenderer-0.9.1-1  vte-common-0.64.2-2  vte3-0.64.2-2  qemu-arch-extra-5.1.0-3

:: qemu 与 qemu-git 有冲突 (qemu-headless)。删除 qemu-git 吗？ [y/N] y
软件包 (2) qemu-git-13:6.0.0.r1992.ge3955ae93f-1 [删除]  qemu-5.0.0-8
```

# 测试失败
原因是一个依赖库（liburing）版本高，每次都是它。。。。。。。

```bash
$ qemu-riscv64 -version
qemu-riscv64 version 5.1.0
Copyright (c) 2003-2020 Fabrice Bellard and the QEMU Project developers
$ qemu-system-riscv64 -version
qemu-system-riscv64: error while loading shared libraries: liburing.so.1: cannot open shared object file: No such file or directory
```
## 解决办法
```bash
$ sudo downgrade liburing
-  5)  liburing    0.7  2  /var/cache/pacman/pkg

$ qemu-system-riscv64 -version
QEMU emulator version 5.1.0
Copyright (c) 2003-2020 Fabrice Bellard and the QEMU Project developers

$ qemu-riscv64 -version
qemu-riscv64 version 5.1.0
Copyright (c) 2003-2020 Fabrice Bellard and the QEMU Project developers
```

最后最后终于是解决了这个问题，感谢兄弟姐妹们的帮助。

closed 2021-06-29 by BackMountainDevil