rcore-os/rCore-Tutorial-v3 #19: 是否应该把默认的编译模式改为debug？

leoleoasd, 2021-06-03:
如题，发现很多地方的默认编译模式是`release`。如：
https://github.com/rcore-os/rCore-Tutorial-v3/blob/ch1/os/Makefile#L3
https://github.com/rcore-os/rCore-Tutorial-v3/blob/ch1/os/Makefile#L46
作为一个教学用的工程，是否应该默认用`debug`模式编译，方便gdb等工具的使用？

或者，qemu模拟器时默认`debug`？

wyfcyx, 2021-06-03:
使用`release`模式主要是希望一些编码不慎导致的UB（未定义行为）能够有更大概率暴露出来，在`debug`模式下很可能可以正常跑，因此并不是主要出于性能的考虑。至于调试的话，目前还不太支持源码级调试，一般打印log信息结合汇编代码级调试就够用了，可以试试目前提供的`make debug`来调试。

leoleoasd, 2021-06-03:
> 使用`release`模式主要是希望一些编码不慎导致的UB（未定义行为）能够有更大概率暴露出来，在`debug`模式下很可能可以正常跑，因此并不是主要出于性能的考虑。至于调试的话，目前还不太支持源码级调试，一般打印log信息结合汇编代码级调试就够用了，可以试试目前提供的`make debug`来调试。

感谢，了解了。

其实我这边macOS折腾了一下源码级别调试，成功了，可以用裸gdb/clion里的gdb链接到qemu虚拟机里调试。需要的话我可以写一下怎么配置到文档里

效果大致如图：
[image]
[image]

wyfcyx, 2021-06-03:
多谢！如果可以的话，请到[文档仓库](https://github.com/rcore-os/rCore-Tutorial-Book-v3)提个PR大概讲讲如何配置调试环境。