rcore-os/rCore-Tutorial-v3 #23: 关于加入汇编代码的适配性的小疑惑

BackMountainDevil, 2021-07-07:
在 [构建用户态执行环境](https://rcore-os.github.io/rCore-Tutorial-Book-v3/chapter1/3-1-mini-rt-usrland.html#id4)中第一次加入了汇编代码。
```rust
#![feature(llvm_asm)]
#![no_std] // 不链接 Rust 标准库
#![no_main] // 禁用所有 Rust 层级的入口点

use core::panic::PanicInfo;

/// 这个函数将在 panic 时被调用
#[panic_handler]
fn panic(_info: &PanicInfo) -> ! {
    loop {}
}

const SYSCALL_EXIT: usize = 93;

fn syscall(id: usize, args: [usize; 3]) -> isize {
    let mut ret: isize;
    unsafe {
        llvm_asm!("ecall"
            : "={x10}" (ret)
            : "{x10}" (args[0]), "{x11}" (args[1]), "{x12}" (args[2]), "{x17}" (id)
            : "memory"
            : "volatile"
        );
    }
    ret
}

pub fn sys_exit(xstate: i32) -> isize {
    syscall(SYSCALL_EXIT, [xstate as usize, 0, 0])
}

#[no_mangle] // 不重整函数名
pub extern "C" fn _start() {
    // loop {}
    sys_exit(9);
}
```

然后用 riscv64 架构可以编译运行

```bash
[kearney@arch os]$ cargo build --target riscv64gc-unknown-none-elf
   Compiling os v0.1.0 (/home/kearney/Documents/code/rust/os-rust/os)
    Finished dev [unoptimized + debuginfo] target(s) in 0.12s
[kearney@arch os]$ qemu-riscv64 target/riscv64gc-unknown-none-elf/debug/os; echo $?
9
```

但是当我尝试用其它架构的时候，编译会报错，如 x86_64 和aarch64,是不是说明这个加入的汇编代码是具有架构指向性的，微机原理学的不咋地。。汇编代码使用指令集中的指令写的，而不同架构的指令集各不相同，因此才有了下面的交叉编译的错误，请教是这样的吗？

```bash
$ cargo build --target x86_64-unknown-linux-gnu
   Compiling os v0.1.0 (/home/kearney/Documents/code/rust/os-rust/os)
error: couldn't allocate output register for constraint '{x10}'
  --> src/main.rs:19:9
   |
19 |         llvm_asm!("ecall"
   |         ^

error: aborting due to previous error

error: could not compile `os`

$ cargo build --target aarch64-unknown-none-softfloat
   Compiling os v0.1.0 (/home/kearney/Documents/code/rust/os-rust/os)
error: unrecognized instruction mnemonic
  --> src/main.rs:19:9
   |
19 |         llvm_asm!("ecall"
   |         ^
   |
note: instantiated into assembly here
  --> <inline asm>:1:2
   |
1  |     ecall
   |     ^

error: aborting due to previous error

error: could not compile `os`
```

wyfcyx, 2021-07-07:
指令`ecall`和通用寄存器名`x10`都是RISC-V架构独有的，因此不能编译到x86或ARM平台上。

BackMountainDevil, 2021-07-07:
十分感谢 @wyfcyx 的回复，解决了心里的小久久

closed 2021-07-07 by BackMountainDevil