rcore-os/rCore-Tutorial-v3 #29: docker make run failed

hitzhangjie, 2021-09-12:
使用项目下的Dockerfile构建镜像成功，运行起来之后/mnt/os目录下make run报错：

```bash
error[E0432]: unresolved imports `embedded_hal::digital::OutputPin`, `embedded_hal::digital::StatefulOutputPin`, `embedded_hal::digital::InputPin`, `embedded_hal::digital::ToggleableOutputPin`
 --> /root/.cargo/git/checkouts/k210-hal-1069b16e3180bc3a/ea121c9/src/gpio.rs:8:29
  |
8 | use embedded_hal::digital::{OutputPin, StatefulOutputPin, InputPin, ToggleableOutputPin};
  |                             ^^^^^^^^^  ^^^^^^^^^^^^^^^^^  ^^^^^^^^  ^^^^^^^^^^^^^^^^^^^ no `ToggleableOutputPin` in `digital`
  |                             |          |                  |
  |                             |          |                  no `InputPin` in `digital`
  |                             |          no `StatefulOutputPin` in `digital`
  |                             no `OutputPin` in `digital`

error[E0432]: unresolved imports `embedded_hal::digital::InputPin`, `embedded_hal::digital::OutputPin`
 --> /root/.cargo/git/checkouts/k210-hal-1069b16e3180bc3a/ea121c9/src/gpiohs.rs:6:29
  |
6 | use embedded_hal::digital::{InputPin, OutputPin};
  |                             ^^^^^^^^  ^^^^^^^^^ no `OutputPin` in `digital`
  |                             |
  |                             no `InputPin` in `digital`

error[E0432]: unresolved import `embedded_hal::prelude`
  --> /root/.cargo/git/checkouts/k210-hal-1069b16e3180bc3a/ea121c9/src/lib.rs:30:27
   |
30 |     pub use embedded_hal::prelude::*;
   |                           ^^^^^^^ could not find `prelude` in `embedded_hal`

error[E0405]: cannot find trait `Read` in module `serial`
   --> /root/.cargo/git/checkouts/k210-hal-1069b16e3180bc3a/ea121c9/src/serial.rs:100:14
    |
100 | impl serial::Read<u8> for Rx<UARTHS> {
    |              ^^^^ not found in `serial`
    |
help: consider importing one of these items
    |
9   | use crate::serial::serial::nb::Read;
    |
9   | use embedded_hal::i2c::blocking::Read;
    |
9   | use embedded_hal::serial::nb::Read;
    |

error[E0405]: cannot find trait `Write` in module `serial`
   --> /root/.cargo/git/checkouts/k210-hal-1069b16e3180bc3a/ea121c9/src/serial.rs:114:14
    |
114 | impl serial::Write<u8> for Tx<UARTHS> {
    |              ^^^^^ not found in `serial`
    |
help: consider importing one of these items
    |
9   | use core::fmt::Write;
    |
9   | use crate::serial::serial::blocking::Write;
    |
9   | use crate::serial::serial::nb::Write;
    |
9   | use crate::stdout::Write;
    |
      and 4 other candidates

error[E0405]: cannot find trait `Read` in module `serial`
   --> /root/.cargo/git/checkouts/k210-hal-1069b16e3180bc3a/ea121c9/src/serial.rs:200:27
    |
200 | impl<UART: UartX> serial::Read<u8> for Rx<UART> {
    |                           ^^^^ not found in `serial`
    |
help: consider importing one of these items
    |
9   | use crate::serial::serial::nb::Read;
    |
9   | use embedded_hal::i2c::blocking::Read;
    |
9   | use embedded_hal::serial::nb::Read;
    |

error[E0405]: cannot find trait `Write` in module `serial`
   --> /root/.cargo/git/checkouts/k210-hal-1069b16e3180bc3a/ea121c9/src/serial.rs:215:27
    |
215 | impl<UART: UartX> serial::Write<u8> for Tx<UART> {
    |                           ^^^^^ not found in `serial`
    |
help: consider importing one of these items
    |
9   | use core::fmt::Write;
    |
9   | use crate::serial::serial::blocking::Write;
    |
9   | use crate::serial::serial::nb::Write;
    |
9   | use crate::stdout::Write;
    |
      and 4 other candidates

error[E0405]: cannot find trait `FullDuplex` in module `embedded_hal::spi`
  --> /root/.cargo/git/checkouts/k210-hal-1069b16e3180bc3a/ea121c9/src/spi.rs:69:25
   |
69 | impl embedded_hal::spi::FullDuplex<u8> for Spi<SPI0> {
   |                         ^^^^^^^^^^ not found in `embedded_hal::spi`
   |
help: consider importing this trait
   |
3  | use embedded_hal::spi::nb::FullDuplex;
   |

error[E0405]: cannot find trait `Write` in module `embedded_hal::serial`
  --> /root/.cargo/git/checkouts/k210-hal-1069b16e3180bc3a/ea121c9/src/stdout.rs:11:30
   |
11 |     T: embedded_hal::serial::Write<u8>,
   |                              ^^^^^ not found in `embedded_hal::serial`
   |
help: consider importing one of these items
   |
2  | use core::fmt::Write;
   |
2  | use embedded_hal::i2c::blocking::Write;
   |
2  | use embedded_hal::serial::blocking::Write;
   |
2  | use embedded_hal::serial::nb::Write;
   |
     and 1 other candidate

Some errors have detailed explanations: E0405, E0432.
For more information about an error, try `rustc --explain E0405`.
error: could not compile `k210-hal` due to 9 previous errors
Makefile:58: recipe for target 'kernel' failed
make: *** [kernel] Error 101
```

easechen, 2021-09-14:
我也同样，在新环境下搭建的
Linux ubuntu 5.4.0-42-generic #46~18.04.1-Ubuntu SMP Fri Jul 10 07:21:24 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
更换了一摸一样的版本还是这个样子。
rustc 1.57.0-nightly (9bb77da74 2021-09-13)

kokoro-aya, 2021-09-15:
我也遇到了同样的问题，一模一样的错误报告。
用的是官方的rCore-Tutorial-v3 的 Ubuntu18.04 镜像，工具链安装一路照着教程做下来的。
会不会是nightly版本不一致？我这边下载下来的版本是rustc 1.57.0-nightly (c3c0f80d6 2021-09-14)，但是教程里又说make run的话会单独下载依赖（就是独立于系统环境里的rustc？）

很迷茫，不知道能不能直接跳过这个tutorial os的步骤直接跟后面的练习和题目，怕后面做到一半又崩了（

wyfcyx, 2021-09-16:
我们已经看到了此问题，正在尝试解决。问题主要是因为在依赖里面使用`embedded-hal`的`1.0.0-alpha.1`版本，不知道为什么cargo使用的却是`1.0.0-alpha.5`版本，目前还不太清楚跟Docker有没有关系。

wyfcyx, 2021-09-16:
如果发现main分支不能跑的话也可以放心按照教程往下走，至少前六章的代码都是能基于Docker正常运行的。

wyfcyx, 2021-09-17:
现在这个问题应该解决了，请尝试下看能否正常运行。 @hitzhangjie  @kokoro-aya  @relaxcn

kokoro-aya, 2021-09-17:
刚把更新后的仓库拉下来试着构建了一下，可以成功进入内核了。
感谢！

closed 2021-09-18 by wyfcyx