rcore-os/rCore-Tutorial-v3 #31: RustSBI panic with QEMU 6.1.0

KernelErr, 2021-10-06:
在QEMU 6.1.0版本上运行项目会出现无效指令错误，经测试使用RustSBI 0.2.0-alpha.6可以解决，是否考虑将bootloader下的文件更新或者在文档中增加提示？
```
[rustsbi] RustSBI version 0.2.0-alpha.1
.______       __    __      _______.___________.  _______..______   __
|   _  \     |  |  |  |    /       |           | /       ||   _  \ |  |
|  |_)  |    |  |  |  |   |   (----`---|  |----`|   (----`|  |_)  ||  |
|      /     |  |  |  |    \   \       |  |      \   \    |   _  < |  |
|  |\  \----.|  `--'  |.----)   |      |  |  .----)   |   |  |_)  ||  |
| _| `._____| \______/ |_______/       |__|  |_______/    |______/ |__|

[rustsbi] Platform: QEMU (Version 0.2.0)
[rustsbi] misa: RV64ACDFIMSU
[rustsbi] mideleg: 0x222
[rustsbi] medeleg: 0xb1ab
[rustsbi-dtb] Hart count: cluster0 with 1 cores
[rustsbi] Kernel entry: 0x80200000
[rustsbi-panic] hart 0 panicked at 'invalid instruction, mepc: 000000008000004c, instruction: 0000000030200073', platform/qemu/src/main.rs:484:17
[rustsbi-panic] system shutdown scheduled due to RustSBI panic
```

wyfcyx, 2021-10-06:
@KernelErr  我们正在迭代新一版的代码，在新版里面是基于RustSBI 0.2.0-alpha.6的。针对这种情况，稍后我在环境配置中简要说明一下吧。

KernelErr, 2021-10-07:
> @KernelErr 我们正在迭代新一版的代码，在新版里面是基于RustSBI 0.2.0-alpha.6的。针对这种情况，稍后我在环境配置中简要说明一下吧。

好的，非常期待 :)

closed 2021-10-07 by KernelErr