rcore-os/rCore-Tutorial-v3 #42: ch4练习求助

lab4ogtc, 2021-12-06:
```
[DEBUG] [kernel] run first task ch4_mmap0
[TRACE] [kernel] trap: scause=Interrupt(SupervisorTimer), stval=0x0
[TRACE] [kernel] run next task ch4_mmap0 with stride=4095
[TRACE] [kernel] trap: scause=Exception(UserEnvCall), stval=0x0
[TRACE] [kernel] syscall: code=222, args=[0x10000000, 0x1000, 0x3]
[DEBUG] MemorySet insert_framed_area start_va:0x10000000, end_va:0x10001000, permission:6
[TRACE] [kernel] find_pte_create vpn=0x10000, pte=0x20150801, level=0
[TRACE] [kernel] find_pte_create vpn=0x10000, pte=0x20150c01, level=1
[TRACE] [kernel] trap: scause=Exception(StorePageFault), stval=0x10000000
[ERROR] [kernel] PageFault in application, bad addr = 0x10000000, bad instruction = 0x80400098, core dumped.
[TRACE] [kernel] find_pte vpn=0x10000, pte=0x20150801, level=0
[TRACE] [kernel] find_pte vpn=0x10000, pte=0x20150c01, level=1
[TRACE] [kernel] find_pte vpn=0x10000, pte=0x20150407, level=2
[DEBUG] [kernel] Task ch4_mmap0 access page 0x10000000: ppn=0x80541, valid=true, readable=true, writable=true, executable=false
[DEBUG] [kernel] write address 0x80541000
```

我在实现mmap方法时遇到应用程序无法访问map后的内存，但在内核中使用对应的memory_set进行软件translate得到的ppn是可以正常访问的，我使用的是DeathWish5/rCore_tutorial_tests下的测试用例，ch4的用例都能正常运行，实在想不出如何来解决，求教各路大神指点迷津。

wyfcyx, 2021-12-06:
请注意在map的时候添加U标志位，不然应用程序无法访问对应的内存。

lab4ogtc, 2021-12-07:
确实是页表项没有添加U标志位，谢谢大神

closed 2021-12-07 by lab4ogtc