rcore-os/rCore-Tutorial-v3 #50: error[E0658]: use of unstable library feature 'asm'

PatrickFFung, 2022-01-20:
I meet an error when I excuted "make run": 
error[E0658]: use of unstable library feature 'asm': inline assembly is not stable enough for use and is subject to change
  --> ~/.cargo/git/checkouts/riscv-ab2abd16c438337b/11d43cf/src/asm.rs:10:23

wyfcyx, 2022-01-20:
@PatrickFFung Just modify the content of `rCore-Tutorial-v3/rust-toolchain` from `nightly-2022-01-01` to `nightly-2022-01-19`. I guess this can solve the problem.

PatrickFFung, 2022-01-21:
It's work. Thank you so much. Another error occurs while Compiling k210-soc v0.1.0 (https://github.com/wyfcyx/k210-soc#09a0742c):
error: cannot find macro `asm` in this scope
  --> src/lang_items.rs:25:5
   |
25 |     asm!("mv {}, s0", out(reg) fp);
   |     ^^^
   |
   = note: consider importing this macro:
           core::arch::asm

error: cannot find macro `asm` in this scope
  --> src/sbi.rs:17:9
   |
17 |         asm!(
   |         ^^^
   |
   = note: consider importing this macro:
           core::arch::asm

error: cannot find macro `global_asm` in this scope
  --> src/trap/mod.rs:26:1
   |
26 | global_asm!(include_str!("trap.S"));
   | ^^^^^^^^^^
   |
   = note: consider importing this macro:
           core::arch::global_asm

error: cannot find macro `asm` in this scope
   --> src/trap/mod.rs:108:9
    |
108 |         asm!(
    |         ^^^
    |
    = note: consider importing this macro:
            core::arch::asm

error: cannot find macro `global_asm` in this scope
 --> src/task/switch.rs:1:1
  |
1 | global_asm!(include_str!("switch.S"));
  | ^^^^^^^^^^
  |
  = note: consider importing this macro:
          core::arch::global_asm

error: cannot find macro `asm` in this scope
   --> src/mm/memory_set.rs:205:13
    |
205 |             asm!("sfence.vma");
    |             ^^^
    |
    = note: consider importing this macro:
            core::arch::asm

error: cannot find macro `global_asm` in this scope
  --> src/main.rs:27:1
   |
27 | global_asm!(include_str!("entry.asm"));
   | ^^^^^^^^^^
   |
   = note: consider importing this macro:
           core::arch::global_asm

error[E0308]: mismatched types
   --> src/trap/mod.rs:107:5
    |
107 | /     unsafe {
108 | |         asm!(
109 | |             "fence.i",
110 | |             "jr {restore_va}",
...   |
115 | |         );
116 | |     }
    | |_____^ expected `!`, found `()`
    |
    = note:   expected type `!`
            found unit type `()`
For more information about this error, try `rustc --explain E0308`.
error: could not compile `os` due to 8 previous errors

wyfcyx, 2022-01-21:
@PatrickFFung There is nothing wrong with `k210-soc`. Now you can pull the latest version of the `main` branch, it should compile.

PatrickFFung, 2022-01-22:
"make run" completed. Thank you so much!

closed 2022-01-22 by PatrickFFung