rcore-os/rCore-Tutorial-v3 #52: [chp 2 & rust] 是否可以将其在cargo工作空间重构？

PeterWrighten, 2022-01-26:
在做chp2的实验时，我尝试将user和os都置于cargo工作空间下，试图重构（非linux，macOS环境下）。但是遇到了如下问题：

1. ```.cargo/config.toml```文件如果分别置于```members```目录（此处为```/os``` 和 ```user```）的话，无法编译通过。但如果将```.cargo/config.toml```置于根目录的话```user```可以编译通过。
2. 当构建工作空间时，user目录下的定义宏```print!/println!```都可以正常使用，但是os目录下的宏却一直报错，在main.rs下报错为“failed to lower macro args to token tree”，但尚且可以识别这个console.rs中定义的宏，而在其他非入口文件，如lang_items.rs和batch.rs，则根本无法识别这些宏。

我的代码在[这里](https://github.com/PeterWrighten/OperatingSystem/tree/chap2/peter_batch_os)，刚入门rust和学习操作系统，望指教。

PeterWrighten, 2022-01-26:
【追加】：
关于宏使用的问题目前已解决。

但batch.rs中的用户栈与内核栈依旧有如下问题：

```rust
static KERNEL_STACK: KernelStack = {    
    data: [0; KERNEL_STACK_SIZE] 
};
```
以上代码rust-analyzer会报错，说无法发现值，不知是rust-analyzer的问题还是代码的问题。
[image]

wyfcyx, 2022-01-26:
rust-analyzer的问题是因为`data`行末需要加一个`,`；至于`config.toml`我觉得不算什么大问题，只要能跑起来就行。或者可以考虑不用`config.toml`，直接在`os/Makefile`里面的cargo命令里面加上对应的设置，不过这样会比较麻烦。

PeterWrighten, 2022-01-26:
> rust-analyzer的问题是因为`data`行末需要加一个`,`；至于`config.toml`我觉得不算什么大问题，只要能跑起来就行。或者可以考虑不用`config.toml`，直接在`os/Makefile`里面的cargo命令里面加上对应的设置，不过这样会比较麻烦。

感谢回复。加了```,```还是会报错orz
Makefile的话，是不是不可以用cargo工作空间那样运行了呢

PeterWrighten, 2022-01-28:
> rust-analyzer的问题是因为`data`行末需要加一个`,`；至于`config.toml`我觉得不算什么大问题，只要能跑起来就行。或者可以考虑不用`config.toml`，直接在`os/Makefile`里面的cargo命令里面加上对应的设置，不过这样会比较麻烦。

目前上述问题均已解决，但是无论是在Linux还是macOS下都出现了无法将应用bin链接到内核的问题。报错如下：
```shell
error: Could not find incbin file '/../target/riscv64gc-unknown-none-elf/release/03priv_inst.bin'
   |
note: instantiated into assembly here
  --> <inline asm>:39:13
```
这只是其中之一，实际上五个应用链接时都会报错反馈错误。

wyfcyx, 2022-01-28:
我对cargo工作空间不太了解，你如果想基于它的话可能需要多尝试一下。比如在`build.rs`里面打印一下当前工作路径，看看链接的时候到底是在哪个路径下。

PeterWrighten, 2022-01-29:
> 我对cargo工作空间不太了解，你如果想基于它的话可能需要多尝试一下。比如在`build.rs`里面打印一下当前工作路径，看看链接的时候到底是在哪个路径下。

好的，我再尝试一下。主要是想熟悉一下Cargo的特性，也因为电脑性能有限不太想跑虚拟机orz
感谢耐心回复

closed 2022-04-03 by PeterWrighten