rcore-os/rCore-Tutorial-v3 #60: ch8 exit实现存在bug

MashPlant, 2022-03-24:
https://rcore-os.github.io/rCore-Tutorial-Book-v3/chapter8/1thread-kernel.html 中说

> 如果进程/主线程先调用了 exit 系统调用来退出，那么整个进程（**包括所属的所有线程**）都会退出

但示例代码显然不能做到终止其他线程的运行，不用运行任何测试就能直接读出来。

当然这也可以用测试来验证。基于commit 9ad3bb43124648b92e2f95673db4a49c56871de2 ，运行如下测试：

```rust
#![no_std]
#![no_main]

#[macro_use]
extern crate user_lib;
extern crate alloc;

use user_lib::{thread_create, exit};
use alloc::vec::Vec;

pub fn thread_a() -> ! {
    for i in 0..1000 { print!("{}", i); }
    exit(1)
}

#[no_mangle]
pub fn main() -> i32 {
    thread_create(thread_a as usize, 0);
    println!("main thread exited.");
    exit(0)
}
```

可能的输出为：

```
>> exit_test
main thread exited.
[kernel] Panicked at src/sync/up.rs:27 already borrowed: BorrowMutError
[kernel] Panicked at src/task/processor.rs:94 called `Option::unwrap()` on a `None` value
[kernel] Panicked at src/task/processor.rs:94 called `Option::unwrap()` on a `None` value
[kernel] Panicked at src/task/processor.rs:94 called `Option::unwrap()` on a `None` value
...
```

虽然这段代码不太符合通常的正确代码的模式，但它也不应该让内核crash。

wyfcyx, 2022-03-24:
Good catch!看起来我忘记在[这里](https://github.com/rcore-os/rCore-Tutorial-v3/blob/ch8/os/src/task/mod.rs#L69)将就绪队列中同进程下的其他线程移除掉了。

MashPlant, 2022-03-24:
一个提示：可能还需要删除timer中的线程。

wyfcyx, 2022-05-21:
目前在ch8上`early_exit`和`early_exit2`两个测例能跑了，ch9和main分支上涉及到I/O阻塞的部分还没改。

chyyuu, 2022-05-22:
看来对于ch9/main branch还需进一步改进