rcore-os/rCore-Tutorial-v3 #86: make run编译错误，提示this struct takes 1 generic argument but 0 generic arguments were supplied

ruiqurm, 2022-08-31:
rustc版本为`rustc 1.64.0-nightly (9a7b7d5e5 2022-07-19)`
在os目录下执行`make run`以后，提示以下错误：
```
error[E0107]: this struct takes 1 generic argument but 0 generic arguments were supplied
  --> src/drivers/block/virtio_blk.rs:18:32
   |
18 |     virtio_blk: UPIntrFreeCell<VirtIOBlk<'static>>,
   |                                ^^^^^^^^^ expected 1 generic argument
   |
note: struct defined here, with 1 generic parameter: `H`
  --> /home/ruiqurm/.cargo/git/checkouts/virtio-drivers-4fdfaa862bcdc399/4ee80e5/src/blk.rs:13:12
   |
13 | pub struct VirtIOBlk<'a, H: Hal> {
   |            ^^^^^^^^^     -
help: add missing generic argument
   |
18 |     virtio_blk: UPIntrFreeCell<VirtIOBlk<'static, H>>,
   |                                                 +++

error[E0107]: this struct takes 1 generic argument but 0 generic arguments were supplied
  --> src/drivers/gpu/mod.rs:19:25
   |
19 |     gpu: UPIntrFreeCell<VirtIOGpu<'static>>,
   |                         ^^^^^^^^^ expected 1 generic argument
   |
note: struct defined here, with 1 generic parameter: `H`
  --> /home/ruiqurm/.cargo/git/checkouts/virtio-drivers-4fdfaa862bcdc399/4ee80e5/src/gpu.rs:15:12
   |
15 | pub struct VirtIOGpu<'a, H: Hal> {
   |            ^^^^^^^^^     -
help: add missing generic argument
   |
19 |     gpu: UPIntrFreeCell<VirtIOGpu<'static, H>>,
   |                                          +++

error[E0107]: this struct takes 1 generic argument but 0 generic arguments were supplied
  --> src/drivers/input/mod.rs:21:35
   |
21 | struct VirtIOINPUT(UPIntrFreeCell<VirtIOInput<'static>>);
   |                                   ^^^^^^^^^^^ expected 1 generic argument
   |
note: struct defined here, with 1 generic parameter: `H`
  --> /home/ruiqurm/.cargo/git/checkouts/virtio-drivers-4fdfaa862bcdc399/4ee80e5/src/input.rs:12:12
   |
12 | pub struct VirtIOInput<'a, H: Hal> {
   |            ^^^^^^^^^^^     -
help: add missing generic argument
   |
21 | struct VirtIOINPUT(UPIntrFreeCell<VirtIOInput<'static, H>>);
   |                                                      +++
```

ruiqurm, 2022-09-01:
看起来是virtio-drivers的问题，这个项目最近加了一个生命周期参数，rcore这边也需要修改一下。但是我用旧版本的virtio-drivers也可以通过编译。

具体的修改：
在os目录下，编辑Cargo.toml文件：
把virtio-drivers加个rev，我这里用的是499338115b7d462f051，“Use a safer approach to the HAL interface (#11)”之前的都可以用。
```
+virtio-drivers = { git = "https://github.com/rcore-os/virtio-drivers",rev = "499338115b7d462f051" }
```

wyfcyx, 2022-09-02:
已经修复了，请更新到最新版本。

closed 2022-09-02 by ruiqurm