rcore-os/rCore-Tutorial-v3 #132: 请教一个问题，ch4中为什么需要修改user test啊？最近遇到的问题可能与这个有关。

Sober7135, 2024-01-05:
https://github.com/rcore-os/rCore-Tutorial-v3/blob/e2c2949d612b737fb93bcf2d7a7d6244e6e397c9/user/src/bin/00power_3.rs#L9C28-L9C28
很久之前的一个commit（[8ee3671269ccbebcc962ef61ec30c3dd525b6d93](https://github.com/rcore-os/rCore-Tutorial-v3/commit/8ee3671269ccbebcc962ef61ec30c3dd525b6d93)），这里将数组变为`static mut`并且放在了全局，~~但是我发现现在的ch4使用原来的版本也能运行~~。

最近我也写ch4，我自己实现的（并没有完全按找教程写）有一个很奇怪的问题。将数组放在全局不会报错，但是放在`main`函数内会。
- 使用debug编译，在数组初始化过程跳到built-in的`memset`处的汇编直接是`unimp`，所以会直接illegal instruction，但是在from_elf函数中我也print了一下，`memset`处的代码并不是`unimp`，是与objdump出来的汇编是一样的
- 使用release，初始化没有问题，但是下方代码中`println`执行1次以上也会illegal instrucion，如果直接注释掉，也能算出答案，并且打印出来。。
[image]

另外，下面这段代码，在release却可以运行。。
```
#[no_mangle]
fn main() -> i32 {
    let mut s: [u64; 100] = [0; 100];
    for i in 1..100 {
        s[i] = i as u64;
        println!("{}", s[i]);
    }
    0
}
```

总之非常疑惑，debug了半天发现了原来ch4的user test改了，我不知道是我写的有问题，还是说出现这些错误是意料之中于是修改了user test（然后后面的commit又修正了这个问题（？））~~由于那个commit太久远，编译不了（大概是因为依赖的git库的依赖太新了），也没办法验证我的想法，所以消耗一下公共资源过来询问一下~~

wyfcyx, 2024-01-05:
数组放在`main`函数里面的话是开在栈上的，全局声明的话是在`.bss`或者`.data`段里面，唯一的区别只有这个。可以看看是不是爆栈了，或者sp寄存器维护出问题了？

wyfcyx, 2024-01-05:
1.`copy_data`确实只需要`&PageTable`就行了，后面改
2.当`self.0`为`usize::MAX`的时候，`self.0+PAGE_SIZE`就溢出了，所以至少这样写还有问题

Sober7135, 2024-01-05:
感谢！

closed 2024-01-05 by Sober7135