rcore-os/rCore-Tutorial-v3 #145: Issues when bumping to at least nightly-2024-02-28

wyfcyx, 2024-02-29:
error:
```Rust
error: the item `log` is imported redundantly
  --> src/logging.rs:7:11
   |
7  | use log::{self, Level, LevelFilter, Log, Metadata, Record};
   |           ^^^^ the item `log` is already defined here
   |
note: the lint level is defined here
  --> src/main.rs:11:9
   |
11 | #![deny(warnings)]
   |         ^^^^^^^^
   = note: `#[deny(unused_imports)]` implied by `#[deny(warnings)]`

error: could not compile `os` (bin "os") due to 1 previous error
```

warning:
```Rust
warning: `/mnt/c/wsl/rCore-Tutorial-v3/os/.cargo/config` is deprecated in favor of `config.toml`
```

LliminM, 2024-04-09:
所以要怎么改正这个错误

wyfcyx, 2024-06-01:
[此提交](https://github.com/rcore-os/rCore-Tutorial-v3/commit/868f00bec1d7270455653b9f71e9868dca49bff1)已成功完成Rust版本升级，故关闭此issue。

closed 2024-06-01 by wyfcyx